mainsail-crew/moonraker-timelapse #119: Parking head cancel pause print, when is timelapse in hyperlapse mode.

ppisl, 2023-07-07:
If you turn on hyperlapse mode and the "Park Toolhead" option at the same time, then TIMELAPSE_TAKE_FRAME is called when the print is manually paused or paused during a filament change, which means the print resumes and continues.

ppisl, 2023-07-07:
I have created simple PR for this issue: https://github.com/mainsail-crew/moonraker-timelapse/pull/120

zellneralex, 2023-07-17:
I need to look in that. First of all I am not sure why you would do a hyperlapse if you need a manual filament change, and also why you do that with park on.
Hyperlapse normally is used to make a picture time based without moving the toolhead, to less interfere the actual print. But anyhow we allowing it.
I will look in your PR most probably at the weekend.

m-abs, 2025-04-29:
I just encountered this.

I enabled hyperlapse, because I had started a print without the modified gcode (accidentally selected the default profile in my slicer) and didn't want to start over.

My Neptune 4 Plus ran out of filament about 3 hours into the 5 hours print and displayed the UI for replacing the filament, but the print had resumed.

I've reverted the setting back to "Layer Macro", as that is my prefered mode anyway.