mainsail-crew/moonraker-timelapse #122: Bug: Snapshot Command Cancelled

eprouty, 2023-07-10:
Near the end of a print a snapshot command was cancelled for an undetermined reason... then it appears to hop between trying to capture the failed frame (1083) and failure + 1 (1084) but instead captures no additional frames after that failure.

Here's the log output showing this event:
```
2023-07-10 07:59:31,203 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001079.jpg) successfully finished
2023-07-10 08:00:23,990 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001080.jpg) successfully finished
2023-07-10 08:01:15,658 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001081.jpg) successfully finished
2023-07-10 08:01:16,810 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001082.jpg) successfully finished
2023-07-10 08:02:09,614 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:02:09,615 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:02:59,673 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:02:59,674 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:03:00,600 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg) cancelled
2023-07-10 08:03:00,600 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg
2023-07-10 08:03:49,298 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:03:49,299 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:04:40,232 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:04:40,232 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:04:40,653 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg) cancelled
2023-07-10 08:04:40,654 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg
2023-07-10 08:05:30,904 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:05:30,904 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:06:17,094 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg) cancelled
2023-07-10 08:06:17,095 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001083.jpg
2023-07-10 08:06:17,426 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg) cancelled
2023-07-10 08:06:17,426 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame001084.jpg
```

Any ideas what might have caused this?

If anyone has a suggestion for where to look in the code base I'd be happy to contribute a fix and/or feature to properly mitigate this issue so that you may lose a frame or two but the timelapse can continue and finish as successfully as possible.

FrYakaTKoP, 2023-07-28:
is it possible that your disk is full?

eprouty, 2023-07-28:
No, it wasn't. Has 30+ GB of free space.

FrYakaTKoP, 2023-07-29:
That's odd! 

could you run following command in the ssh terminal a couple times and post the output of it:
```wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/test.jpg```

Also can you write more details about your setup: Image used (like MainsailOS, raspiOS, Armbian), streamer software (like crowsnest, bare mjpgstreamer)

Also why you have a extra subfolder "printer" in your frames folder? Do you use Multiinstance?

ElectroDiscover, 2025-01-22:
Hello, I have been running into same issue, output is same and when mannually running wget command it works. I use old android phone as camera over ip adress, it has snapshot feature so no clue what could be the problem.