mainsail-crew/moonraker-timelapse #123: is there a way to change the default rendering frame from 2 to 10? thanks

drumstick77, 2023-07-20:


FrYakaTKoP, 2023-07-28:
sorry can you clearfy what do you mean with "default rendering frame"? Only thing i can think of whould be rendering Threads, which is hardcoded to 2 atm.

drumstick77, 2023-07-28:
when you click on the timelapse option on the left within moonraker page, if you've just done one print with timelapse, to right of that screen, you can normally see an option if we want to re-render it using different frame (hence longer timelapse video).
Currently the default number of frame is two, so that's what got generated automatically. What I'm trying to know is if there's a way to change that from 2 to 10 frames by default. Hope that's clear. cheers

FrYakaTKoP, 2023-07-29:
ok that's easy, just go to the settings menu of your webinterface and change the "Output Framerate" to your desired value. 

In Mainsail:
[image]
[image]
[image]

in Fluidd:
[image]
[image]
[image]

Strange is that your default is 2 fps, normally it should be 30.  If changing it in the webui doesn't work i would need a moonraker.log to see what's wrong. Most likely you have defined `output_framerate:` in your moonraker.conf

i'm closing this issue, feel free to reopen it if your problem persists or have additional questions to that topic

closed 2023-07-29 by FrYakaTKoP

drumstick77, 2023-07-29:
AWESOME! So easy and yet I've missed it! Cheers!