mainsail-crew/moonraker-timelapse #127: [Makefile:20: install] Error 100

sdw03262, 2023-07-29:
I have included a screen shot of the error. it preventing me from installing timelapse on my printer.

[image]

sdw03262, 2023-08-01:
I am fairly new to python. any chance you could point me in the direction of a work around or solution for that issue?

sdw03262, 2023-08-01:
Thank you for the help

sdw03262, 2023-08-01:
@mryel00 thanks for the help. I had to update the time and date with timedatectl set-ntp yes but it appears to have cleared up the issue.

mryel00, 2023-08-01:
That's good. I'm glad that I could help you.

The crowsnest part was just added in case you want to use crowsnest for your camera stream.

closed 2023-08-01 by zellneralex