mainsail-crew/moonraker-timelapse #128: Repo has untracked source files: ['moonraker/components/timelapse.py'] error after update.

Talha909, 2023-07-31:
I got this error after udpate. After that i reinstall the same but the error was their. 
[image]
[logs-20230701-190902.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/12215906/logs-20230701-190902.zip)

miggleswort, 2023-07-31:
Same error for me today. I deleted the symlink to timelapse.py from moonraker/components/ and now the Timelapse menu is gone in both Settings and the main menu.

Grantsauto, 2023-07-31:
Same error here as well. I removed and reinstalled the package while the errors remain.

ayesim, 2023-08-01:
same error.

Delanjp949, 2023-08-01:
Smae here

Winky1st, 2023-08-01:
same here - sry for bump
cheers

zellneralex, 2023-08-01:
Not an error at all, the messages tells you that you did install a non standard moonraker file called timelapse

ayesim, 2023-08-01:
then how should we install timelapse? what is the correct way?

zellneralex, 2023-08-01:
You did it the correct way, as said that is an information, reminder. When you need to do a hard or soft reset of moonraker that file would be gone.

Delanjp949, 2023-08-02:
No error in timelapse,  No soft or hard recovery option shown as no erro detected by Mooraker.  Yet the "moonraker/components/timelapse.py" error is still reported

zellneralex, 2023-08-02:
I know reading is really hard work!!! As said now 2 times the message is not an error it is only an information.

espooblue, 2023-08-02:
Same error Here as well.

KiwiChrist, 2023-08-02:
It does read like an error, however it's itention (as I was told) is informative only. The printer works as it should.

No one likes change they do not understand, until they educate themselves on what the change is, whether good or bad, improvement or regression.  

My initial response to the new informative nofication was ... "oh no,  what have I done, how do I fix this now" and panic set in. 
I feel the information presented could be improved upon.  Example: I would definitely not have reacted or an issue if it read somehting like this... 
"This Repository has unofficial plugins or addons installed. Please be aware that these unofficial plugins may cause problems with updates.  Unofficial Repo files: ['klippy/extras/z_calibration.py']".  This is more of an informative warning rather than...
"Repo has untracked source files: ['klippy/extras/z_calibration.py']", which does not explain to an end user really anything other than a "Warning, something is wrong".

meteyou, 2023-08-03:
@espooblue did you read the answers of this ticket? and why do you call it error? where do you see it in the interface, that is is an error?

espooblue, 2023-08-03:
> @espooblue @Delanjp949 @Talha909
> 
> did you read the answers of this ticket? and why do you call it error? where do you see it in the interface, that it is an error?

is there someway to hide or fix it or correct it or something ......or just leave it as is??

meteyou, 2023-08-03:
@espooblue can you pls answer my questions? what do you want to fix or correct on a piece of information?

zellneralex, 2023-08-03:
For all here as soon you had installed timelapse you would have seen that
[image]
If you would have taken the time to run a bit Status in the moonraker folder

locki-cz, 2023-08-04:
> For all here as soon you had installed timelapse you would have seen that [image] If you would have taken the time to run a git status in the moonraker folder.
> 
> And that information is no presented to you in the Updater …

how to fix it?

meteyou, 2023-08-05:
@locki-cz do you read the text?

zellneralex, 2023-08-05:
@locki-cz delete timelapse!!! That is the only fix. Sorry to say we tried to explain now more than once that this is the **normal and expected** behavior. That is the only way as long neither Klipper nor moonraker have a official plugin system .

TonyK54, 2023-08-05:
I deleted 'moonraker/components/timelapse.py' but I still see the blue information dot. Restarted moonraker, still there. Restarted the host and still there. Thoughts or just ignore?

zellneralex, 2023-08-05:
The check runs only once a week or month depending on the setting of your moonrakrr.conf. You can manual check for updates by the cycle symbol in the head row of your updater section.

KiwiChrist, 2023-08-05:
@TonyK54 You may need to clear the browser cache to update and clear what is displaying.

meteyou, 2023-08-06:
This text is only cached in moonraker, as long, as you don't refresh the update manager (button on the right top of the update manager, like alex wrote above).

neclimdul, 2023-08-06:
It seems like from the discussion this is everything "working as intended" but I think from the number of people here and my own experience the experience isn't the best. "its informational not an error" but its a nag that feels like it should be fixed regardless of it being "information"

1. It stands out visually like a warning. 
2. If its untracked, how does it get updated? Sounds bad in the update section. Something is clear wrong right?
3. Experience with git says untracked files are bad. Add them and track them or they'll be lost or out of sync.
  - If you read the git information from the command line, it re-enforces this.  `use "git add" to track` but we're using git as a read only deployment tool so this isn't right? right?

So, probably something to be done to make this more user friendly. In other projects I work on that have a directory that will contain known external files you usually add a `.gitignore` that avoids these sorts of messages. There are project files in this directory so it'd have to be a bit clever but that might work. Or maybe the updater could provide more information that could be more informative that rather then a raw git message saying "there's this component installed." ? Or maybe just the timelapse file contains a docblock explaining why its there and why its not tracked and how to manage it?

zellneralex, 2023-08-06:
That is clearly out of control of the Timelapse plugin Devs. It is the decision of the moonraker dev. 

Currently there is no other way for addons like Timelapse to copy (or link) there python file in the component folder. And all these files are per default untracked files. But for Timelapse there are tracked and updated with there own repo. That’s the reason we only link the file in the component folder of moonraker.

So to make it short, that’s currently the only way and you are the minority of users.

zellneralex, 2023-08-06:
But to clarify Mainsail never uses the Main color as Warning. Warning are always orange like 
[image]

neclimdul, 2023-08-06:
I'm not sure how you determine metrics but like 10 people reporting the issue in a week where a small subset of users are going going to take the time to report it seems like a lot. 

Anyways, I was just trying to engage in making this interface better which is why I tried to provide some suggestions on how moonraker and this project might be able to improve things. 

I understand this project is not _directly_ in control of this the moonraker files but assumed the suggestion might be a starting point for engaging with moonraker as this community probably has more influence then some rando from the internet. That's also why I suggested the docbloc which this project _could_ provide. 

As for the docblock I was thinking something like this that would further the discussion on the moonraker project where a deeper fix/improvement could be implemented.

```
# This file is linked in the moonraker component directory by the
# moonraker-timelapse project to provide additional functionality. It may show
# as dirty or untracked in your update manager. This is a limitation of how
# project are able to extend moonraker.
#
# See https://github.com/Arksine/moonraker/issues/dne_yet for discussion on
# improving this interaction.
```

zellneralex, 2023-08-06:
10 out of more that 11k downloads of Mainsail 2.6.2. And as 2.6.2 is out since that issue it started you can scale it by yourself.
And to insure there are more that 50k downloads of the latest MainsailOS so that I am sure that the majority of the 11k have that info symbol.

I bet 9 out of that 10 did not even open the file so an infoblock in the file header would not help at all. I know what I am talking about as I did the timelapse macros and mainsail config macros that you have in your config folder. And helping users clearly show that even that files that you can open within the UI are not opened by many …

We discussing the wish/need of an (let’s call it) plug in system for more than 2 years now, there where simple more important stuff on the plate of Arksine.

And as stated now many times, the message above is an info that should remind you to be careful if you do a hard or soft restore as than the file (better say the link) will be missing. It should also remind you about all the other modules you might copy or link in moonraker or Klipper.

neclimdul, 2023-08-06:
OK well if my suggestions aren't helping and the project team doesn't see this as an issue maybe close the issue?

zellneralex, 2023-08-06:
I will soon, I only want to insure that the next one is not opened the next day. So we decided to let this open for a while.

Do not get me wrong, we appreciate feedback, but in that particular case I am not sure if anything will help.

what is on the todo list of Mainsail is to PR moonraker to get a more helpful messages for all the types of info and warnings that is generated for the updater.
But we still in the discussion internal how to improve the presenting of these infos and warnings.

meteyou, 2023-08-06:
@neclimdul 10 people in this time is quite "nothing"... and the fact, that we don't close this ticket, means that we have to change/adapt something.

But starting a panic at all ends is the wrong thing to do now... what bothers me is more that users come here and still comment on it "i have the same *error*! how can i fix it?". a contribution to the issue, on the other hand, as you are doing, i find very welcome!

unfortunately, my current week was a horror and therefore I had little time to focus on this issue...

my current plan is that the message will always be displayed with the "disabled" color (gray). this way we can avoid that users use the primary color red and it looks even more like an "error". i have already prepared this, but have not yet done a PR in mainsail.

as a second and more important step, i want to do a PR on Moonraker to make the text much more understandable to non-git people. the current wording is extremely git people oriented, which unfortunately a lot of the Klipper/Moonraker/Mainsail people are not. another user has already sent me a "template" for this. but what i want to add as a good addition from your posts is the "update capability". this is not taken into account in the current text i have. in the case of the timelapse plugin, you could also write "if it is a plugin, this is usually updated separately via its own update manager entry." this i would still have to add.

neclimdul, 2023-08-06:
@meteyou Sorry, I wasn't trying to start a panic, only trying engage in helping with a fix. I understand my comments have been "active" but just that I had some time to engage with this issue and have been trying to provide suggestions to help move things forward since the existing comments from maintainers seemed to be fairly dismissive including the direct responses to my suggestions.

I think your plan seems good and I totally understand juggling time on getting things done in open source. Thanks for providing your plan for moving this forward. Let me know how I can help and stop compulsively checking this status box to see if there's something I need to fix.

meteyou, 2023-08-06:
@neclimdul I am very thankful for your input. your "suggestion" about the update capability has certainly moved us one step closer to a good solution! the dismissal here in the history should rather apply to the posts "i have the same error". unfortunately, these are not very helpful, but rather demotivate...

thx for your time and i will post all PRs that deal with this issue are cross-linked here.

peperonikiller, 2023-09-09:
Came here thinking I had ran into an issue somewhere.  Sorry to post and not contribute but I thought it was an error message as well. Don't use command line GIT ever so w/e.

SckyzO, 2023-09-09:
This error is because you had use moonraker-timelapse.
Juste remove file (or move into /tmp), and this error disappears.
After that, if your timelapse doesn't work, reconfigure it.

meteyou, 2023-09-09:
@SckyzO first of all, this is NO error. there is never the word "error" in the message or it will not displayed like an error (red or something like that).

if you remove this file or move it into /tmp, you will uninstall the plugin at the same time. this file have to be on this place to work fine.

pls read our messages above! this is only an information, that there is this file in the moonraker directory and it is untracked (because its not part of the official moonraker repo).

KiwiChrist, 2023-11-20:
The fix for me was to uninstall Moonraker via KIAUH. THe warning then went away.

meteyou, 2023-11-20:
> The fix for me was to uninstall Moonraker via KIAUH. The warning then went away.

this is no fix. you just removed your timelapse plugin with this step. pls read the rest of this conversation. this is NO warning! its just an information!!

closed 2023-11-20 by meteyou

AGrahazl, 2023-12-31:
after reading this whole thread,  WOW,  just wow.   And yes I too am now seeing this warning, info banner, non error thingy .  For the lay person that sees this type of info banner, the first thought is somethings broke, some nefarious code insertion, end of the world type code.  Now we get it, its just info and not a problem.  Solution; just uninstall timelapse or just ignore and enjoy the pretty orange colors of the banner and little i icons thingy.  Got it.

Alternative:  Yep we understand your concern.  It does not seem break timelapse though.  This is simply a result of not having a viable plugin upstream within Mainsail.  Please visit the Mainsail site and request or upvote request for plugin.  This will help correct things that are triggering this "information" button/orange notification banners and make for a better experience.  Help us help you.

There one and done

meteyou, 2023-12-31:
> Alternative: Yep we understand your concern. It does not seem break timelapse though. This is simply a result of not having a viable plugin upstream within Mainsail. Please visit the Mainsail site [...]

The warning comes from Moonraker and the time lapse plugin is in Moonraker. So a mainsail plugin system would change nothing...

AGrahazl, 2024-01-01:
Yep, my bad,  I should of wrote moonraker and not mainsail.  Excellent
attention to detail though, but in the end I think you very much get my
point.  Help us help you.  On the other hand the snarkiness is entertaining.

BelgarionNL, 2024-01-01:
perhaps its time to create a new mainsailOS image since the latest I used today got a bunch of "errors" for the newbies. And simply bashing on everyone in this channel isnt really helpful because if you dont understand you dont understand....

meteyou, 2024-01-01:
@BelgarionNL without any description, I don't know what you mean... Do you speak about this? https://docs-os.mainsail.xyz/getting-started/first-boot#updates

zellneralex, 2024-01-01:
@AGrahazl it might be time to update your Mainsail also …
[image]
It looks like that now for the last 2 to 3 versions

zellneralex, 2024-01-01:
@BelgarionNL no one is bashing here, but if after an answer that explains the situation the next one post the same comment feels like a real waist of our time and energy.

Remember we doing that all in our spare time, it is absolutely free for you.

BelgarionNL, 2024-01-08:
@zellneralex well I could piece things together but I can understand why some cant. it wasn't the clearest of answers. But water under the bridge.

Please don't start with the whole old martyrdom routine with: its free so you shouldn't be critical etc. etc. 

I for one am VERY aware of everyone's time commitment to make it better for yourselves and others and it's without question you guys rock.

meteyou, 2024-02-09:
@chuckkay messages you should read, will be in a different color and the update button will also change the color!

So this post will be the last message here, and I will lock this issue because every post here is only a waste of time from the crew! This message is information like a layer count, a print progress % or anything else. It's only information; we will display it as information and not change it. Also, when Moonraker has a real plugin system, in the same place, we will display that there are plugins installed and you have to pay attention to these plugins, that every update can break these plugins or the plugins can break Moonraker.

So, the last statement about this is: We will not change the message on this place. It is information, and in some time, you will need this information!

I had a personal experience yesterday: I had to reset Klipper & Moonraker because I had a lot of changes for new programming in Mainsail. With this information, I could check which plugins I had to reinstall after a reset.