mainsail-crew/moonraker-timelapse #129: timelapse not rendering after print job

remizca, 2023-08-09:
as title says, ever since updating via klipper, i noticed that it kept failing and there are no timelpase records anymore. what should i check? camera works fine on main fluidd page.

ptegler, 2023-08-09:
noticed this too after updates...back on 7/29 was last successful render.
all frames (no sequence number skipped) were written to /tmp/timelapse, but all frames have zero data (less than 4K file sizes)
command line attempt at rendering (command captured from moonraker.log error)
fails as frame data is invalid. this is with the newer V4 webrtc config

FrYakaTKoP, 2023-08-10:
make sure the snapshot url which is configured for your camera is a really a still image not a stream (verify it with your browser by entering the same address). This is a common issue people have and will result in exactly that error picture (frames get collected with only header data in it, not real picture data).

If you can verify that the snapshoturl is pointing a still jpg image and still have a problem, please upload your moonraker.log (with a timelapse print in it) and maybe a sample frame so i can analyze those.

ptegler, 2023-08-10:
yes, realized an error for the webcam selected....  webrtc recent updates (crowsnest v4 updates)  url snapshot script was wrong.
Have not tested (run a print) but a browser url test returned a frame image as expected.

But there is still the issue with linear motion vibration not writing out the graphs

remizca, 2023-08-15:
hello. sorry for replying just now. where do i find the setting for the snapshot url in fluidd?

kwotattoos, 2023-12-20:
> noticed this too after updates...back on 7/29 was last successful render. all frames (no sequence number skipped) were written to /tmp/timelapse, but all frames have zero data (less than 4K file sizes) command line attempt at rendering (command captured from moonraker.log error) fails as frame data is invalid. this is with the newer V4 webrtc config
> 
> update.... further testing/checking.... timelapse shows under moonraker update manager entry as an 'untracked' source file Perhaps related.... klipper update manager shows linear_movement as an untracked' source file and does not properly write out files as well

did you ever get this figured out? my timelapse is rendering and gives me a thumbnail but no video. i can save frames and it has all the layer snapshots so i know it taking the pictures. i also have the same error about update manager saying timelapse.py is an untracked source

remizca, 2023-12-23:
> did you ever get this figured out? my timelapse is rendering and gives me a thumbnail but no video. i can save frames and it has all the layer snapshots so i know it taking the pictures. i also have [...]

i ended up reinstalling it. works fine now but i still can't find the proper tuning needed to get rid of the blobs whenever i use the timelapse and let the print head park away from the print. better timelapses i have are when the printhead stays at the center when taking the shots.

Dogfight84, 2023-12-26:
Hi there, i've got the same issue with `mjpeg` codec version of `timelapse.py` and i managed to have an output video file playable from windows pc, just changing output extension from .mp4 to .avi, in all mentioned part of `timelapse.py` file:
```c
            # build shell command
            cmd = self.ffmpeg_binary_path \
                + " -r " + str(fps) \
                + " -i '" + inputfiles + "'" \
                + filterParam \
                + " -threads 2 -g 5" \
                + " -vcodec mjpeg" \
                + " -pix_fmt " + self.config['pixelformat'] \
                + " -b:v 7M" \
                + " -an" \
                + " " + self.config['extraoutputparams'] \
                + " '" + self.temp_dir + outfile + ".avi' -y"
``` 
```c
            # check success
            if cmdstatus:
                status = "success"
                msg = f"Rendering Video successful: {outfile}.avi"
                result.update({
                    'filename': f"{outfile}.avi",
                    'printfile': gcodefilename
                })
                # result.pop("framecount")
                result.pop("settings")

                # move finished output file to output directory
                try:
                    shutil.move(self.temp_dir + outfile + ".avi",
                                self.out_dir + outfile + ".avi")
                except OSError as err:
                    logging.info(f"moving output file failed: {err}")
``` 
Please take care about followings:
- do not consider any matter of quality and ecc...i was just finding a solution to this bad issue.
- by the way, fluidd player is not reconizing the file anymore from timelapse menu, so it will prompt you to download the video file,

Here below the previous output file not working on PC, the corrected file in .AVI format it's not uploadable due to github platform policy.

https://github.com/mainsail-crew/moonraker-timelapse/assets/89547165/bbeafb4c-6ff7-439f-b486-5d0146da7f9a

austwhite, 2024-03-06:
I have to ask.  Is this issue only with MJPEG codec or have others had it with different CODEC's ?
I was about to change to a setup that uses a camera that needs MJPEG.   The current setup is 100% successful with timelapse and I will delay the change if this issue is affecting MJPEG only.

Dogfight84, 2024-03-07:
> I have to ask. Is this issue only with MJPEG codec or have others had it with different CODEC's ? I was about to change to a setup that uses a camera that needs MJPEG. The current setup is 100% successful with timelapse and I will delay the change if this issue is affecting MJPEG only.

I'm not sure if i understod your reply, but yes only with MJPEG codec because i managed to have only this codec working in my Creality K1.
My two beers are in fridge still waiting for @FrYakaTKoP reply on my issue.... ;)

teamfighter, 2024-03-24:
Subscribed, same  problem with my k1 max

LouSassel, 2025-02-22:
> make sure the snapshot url which is configured for your camera is a really a still image not a stream (verify it with your browser by entering the same address). This is a common issue people have [...]

That was 100% my issue! I had the same symptoms as everyone else and this fixed it.