mainsail-crew/moonraker-timelapse #130: moonraker timelapse is generating a jpg with an unknown file format

Mstar3D, 2023-08-28:
My webcam is working and I installed the moonraker timelapse plug in. A zip file is generated after the print with jpg images of every layer. The jpg's are in an unknown file format (Ican not open them). They also can't be viewed in fluidd.

dev-w0olf, 2023-09-24:
Got the same problem :(

dimasmith, 2024-04-10:
Got the same issue. Fixed by changing the `snapshoturl` property in the `moonraker.conf` file to `http://localhost:4408/webcam/?action=snapshot` (notice the `webcam` part).

The cause of the issue, in my case, was an incorrect URL. Generated jpg files actually had an HTML code inside, instead of image data. All due to the wrong default settings.

Hope that helps.

johnbeer13, 2024-04-12:
Всем привет!
У меня была аналогичная проблема, пока я не дополнил файл `moonraker.conf` в `[timelapse]` строкой
`snapshoturl: /webcam/?action=snapshot`