mainsail-crew/moonraker-timelapse #135: Timelaps not rendering

miklas206, 2023-10-05:
So i have an error where i cannot render my timelapse its just stuck with this like the picture i have tried to search the web for help but could not find anything.....

[image]

LeoMGames, 2024-01-01:
I have the same problem. My workaround is to download the images and use this tool. https://timelapsetool.com/

kjkent, 2024-01-23:
Yep I have this issue too -- the timelapse successfully rendered exactly 10 times, then hasn't worked since. grepping through the logs shows `timelapse.py:render() is already running` a bunch of times, and this persists across reboots & restarts of the systemd services.

Interestingly, the last successful render kicked out this error message that I missed at the time because the video did render:

```log
./moonraker.log.2024-01-21:2024-01-21 21:51:16,278 [timelapse.py:render()] - start FFMPEG: /usr/bin/ffmpeg -r 30 -i '/tmp/timelapse/printer/frame%6d.jpg' -threads 2 -g 5 -crf 17 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/printer/timelapse_CE3S1_galile0's_Label_Slot_Bin_1x2x6_Gridfinity.gcode_20240121_2151.mp4' -y
./moonraker.log.2024-01-21:2024-01-21 21:51:16,279 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7f3a48897b10>>, <Task finished name='Task-560220' coro=<Timelapse.render() done, defined at /home/kjkent/moonraker/moonraker/components/timelapse.py:598> exception=ValueError('No closing quotation')>)
./moonraker.log.2024-01-21-Traceback (most recent call last):
./moonraker.log.2024-01-21-  File "/home/kjkent/moonraker-env/lib/python3.11/site-packages/tornado/ioloop.py", line 750, in _run_callback
./moonraker.log.2024-01-21-    ret = callback()
./moonraker.log.2024-01-21-          ^^^^^^^^^^
./moonraker.log.2024-01-21-  File "/home/kjkent/moonraker-env/lib/python3.11/site-packages/tornado/ioloop.py", line 774, in _discard_future_result
./moonraker.log.2024-01-21-    future.result()
./moonraker.log.2024-01-21:  File "/home/kjkent/moonraker/moonraker/components/timelapse.py", line 711, in render
./moonraker.log.2024-01-21-    scmd = shell_cmd.build_shell_command(cmd, self.ffmpeg_cb)
./moonraker.log.2024-01-21-           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
./moonraker.log.2024-01-21-  File "/home/kjkent/moonraker/moonraker/components/shell_command.py", line 368, in build_shell_command
```

aredington, 2025-01-17:
I got bit by this, I believe it may originate at https://github.com/mainsail-crew/moonraker-timelapse/blob/main/component/timelapse.py#L630 where the filename of the originating gcode is interpolated.

The filename I printed when I got bit was "joe's-synth-stand.gcode" and problem is related to the filename including a single tick character.

elijalukas, 2025-08-15:
anybody found a solution? i have the same issue and no special file name.