mainsail-crew/moonraker-timelapse #139: Need to Rotate TIMELAPSE

TheVoronModder, 2023-10-24:
I have a ArduCam 518, I set the camera to the correct rotation 180° so its right side up, the camera is installed upside down FYI.

I cannot find a way to get the timelapse to be upright all the videos are upside down....

How do I fix this?

skyeshadow94, 2023-10-25:
On Minsial it is under the webcams setting interface for me atleast

arcadien, 2024-06-08:
Hello, in mainsail the webcam view is ok, but the timelapse is not rotated. 
Moonraker log seems to trace correct commands, including `-vf 'hflip,vflip'`

```
[...]
1766 2024-06-08 20:57:20,073 [shell_command.py:_check_proc_success()] - Command (wget http://127.0.0.1:8080/snapshot -O /tmp/timelapse/frame000494.jpg) successfully finished
2024-06-08 20:57:36,066 [timelapse.py:render()] - start FFMPEG: /usr/bin/ffmpeg -r 30 -i '/tmp/timelapse/frame%6d.jpg' -vf 'hflip,vflip' -threads 2 -g 5 -crf 23 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/timelapse_Cam and light support(2)_PLA_2h58m.gcode_20240608_2057.mp4' -y
2024-06-08 21:01:40,282 [timelapse.py:render()] - Rotate preview image cmd: /usr/bin/ffmpeg -i '/home/pi/timelapse/timelapse_Cam and light support(2)_PLA_2h58m.gcode_20240608_2057.jpg' -vf 'hflip,vflip' -an  '/home/pi/timelapse/timelapse_Cam and light support(2)_PLA_2h58m.gcode_20240608_2057.jpg' -y

```
[image]

arcadien, 2024-06-08:
After some more investigation. When webcam are configured to be rotated in Mainsail, the frames are captured rotated (as in Mainsail views). If ffmpeg rotates _again_ frames, then they are back to bad format.

arcadien, 2024-06-09:
After some more investigation. I have two cameras. First one is not rotated (std webcam) the second is (Raspberry cam).
The rotation param seems to apply to both when using timelapse.