mainsail-crew/moonraker-timelapse #144: Timelapse crashes when trying to update settings

Compupaq, 2024-01-01:
When I try to update the settings in Mainsail, (settings, timelapse tab), if I delete everything in a field to type in a different value, timelapse crashes and unloads.  It'll continue to not work until I manually define the broken value in the timelapse section in moonraker.conf.

This is the message I got in the moonraker.log after deleting everything in the park travel speed field:

`2024-01-01 13:11:50,195 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: machine.timelapse.post_settings, Code: 400, Message: Unable to convert argument [park_travel_speed] to <class 'int'>: value recieved: `

It seems like the settings tries to update on every keypress, which causes the error once the field is blank.

zellneralex, 2024-01-01:
@meteyou

zellneralex, 2024-01-01:
@Compupaq Imwill try to reproduce that.

Compupaq, 2024-01-01:
I forgot to mention, I'm also running the current versions of everything on a Raspberry PI 4b running Armbian.
[image]

meteyou, 2024-01-01:
just a quick side question. why are you using armbian on the raspberry pi? where do you see the advantage compared to raspberry pi os?

Compupaq, 2024-01-01:
Just trying it out to see how it works.  It's been running fine so far 🤞