mainsail-crew/moonraker-timelapse #145: Init. Moonraker: timelapse An error was detected during initialization the moonraker component 'timelapse'. Please check the log file and fix the issue.

XJRpilot, 2024-01-02:
hi im trying to get the timelapse to work. here is my error,
Init. Moonraker: timelapse
An error was detected during initialization the moonraker component 'timelapse'. Please check the log file and fix the issue.

and here im my moonraker log
[kevs moonraker log.log](https://github.com/mainsail-crew/moonraker-timelapse/files/13807142/kevs.moonraker.log.log)

please any help would be greatly appreciated

XJRpilot, 2024-01-02:
ive done a fresh install of ubuntu 22.04.3LTS and cannot make any adjustments to the settings without it crashing and timelapse tab disappearing from the menu. Reboot to make it reappear.

XJRpilot, 2024-01-03:
Hi 
today i reinstalled fresh ubuntu 22.04.3 LTS and reinstalled all components required, then followed the install and config steps including these

        - for Printer E5P:
        [timelapse]
        output_path: ~/E5P_data/timelapse/
        frame_path: /tmp/timelapse/E5P

        - for Printer E3MN:
        [timelapse]
        output_path: ~/E3MN_data/timelapse/
        frame_path: /tmp/timelapse/E3MN
it has given me a new warning

Moonraker warning
Unparsed config option 'ooutput_path: ~/E5P_data/timelapse/' detected in section [timelapse]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error

when i try to change any settings it crashes and disappears from the interface settings. only a reboot brings it back.

i also did an install on a fresh pi4 using mainsail/klipper and had the same issue,
reinstalled a RPI OS and same errors again

[kevs moonraker log.log](https://github.com/mainsail-crew/moonraker-timelapse/files/13816655/kevs.moonraker.log.log)

thanks

VellD, 2024-01-13:
there was such a problem.
reinstalled moonraker-timelapse and the timelapse menu appeared and everything works fine
I just entered it in the terminal:
```
cd ~/moonraker-timelapse
make install
```

XJRpilot, 2024-01-14:
https://github.com/mainsail-crew/moonraker-timelapse/assets/145925264/80f508a4-160a-4ebf-95a9-80b24b5b324a

i followed the instructions and reinstalled moonraker-timelapse and the fault exists. i have 3 fresh installs on 3 different RPi's and have the same issue with all three

XJRpilot, 2024-01-14:
i did another fresh install. the minute i attempt to change the settings after selecting park head, the time-lapse menu collapses and dissapears, it takes a reboot to make it reappear. As per the video above.
Here are the logs
[kevs logs-20240115-071809.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/13933156/kevs.logs-20240115-071809.zip)