mainsail-crew/moonraker-timelapse #146: Timelapse errors out with "Error taking timelapse frame" after approximately 650 frames

DeQuosaek, 2024-01-04:
I'm using the timelapse feature to take time-lapse videos of my 3D prints in Klipper on a Neptune 4 Pro printer and every time my time-lapse gets to around 650 frames it returns the error "Error taking timelapse frame".  

The moonraker log shows the following errors and a return code of 3.  I'm assuming that might mean I've run out of drive space.
[shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000661.jpg) exited with return code 3

zellneralex, 2024-01-09:
If your sd card runs out of space you need a bigger one, or remove stuff you do not need

DeQuosaek, 2024-01-09:
This might not be a Moonraker problem then. I'll have to look around and
figure out how to mount the SD card and point the output to that location.
I'm an inexperienced intermediate user with Linux. My apologies for raising
an issue that probably isn't related to the code.

zellneralex, 2024-01-09:
You can define the output and the frame path inside your moonraker.conf 
```ini
[timelapse]
output_path: ~/printer_data/timelapse/
frame_path: /tmp/timelapse/
```

DeQuosaek, 2024-01-09:
Thank you for the information.  Unfortunately I'm running on a Neptune 4 Pro and the TF card slot doesn't work for this purpose and the USB is taken up by my webcam.  I'll have to figure out a workaround or just load everything onto a separate Raspberry Pi and take the processing away from the small printer board.  Thanks again!

DeQuosaek, 2024-01-09:
Closing issue, as this does not appear to be an issue with the code.

closed 2024-01-09 by DeQuosaek

zellneralex, 2024-01-09:
Maybe reduce the resolution of the cam as that would end up in smaller snapshot files.

DeQuosaek, 2024-04-08:
> You can define the output and the frame path inside your moonraker.conf
> 
> ```ini
> [timelapse]
> output_path: ~/printer_data/timelapse/
> frame_path: /tmp/timelapse/
> ```

Thanks for the information on the default location of the frames!  Until I figure out a better solution to the storage issue, at least I can move already taken frames out of there to make room for more while the time-lapse is going.