mainsail-crew/moonraker-timelapse #147: conflicted with klipper_tmc_autotune

Rojer-x86, 2024-01-06:
this plugin conflicted with plugin klipper_tmc_autotune

after installation the plugin worked
but after reboot system klipper crashes on startup with error: 
Internal error on command:"HYPERLAPSE"
Internal error on command:"UPDATE_DELAYED_GCODE"

log file:
[klippy-timelapse.log](https://github.com/mainsail-crew/moonraker-timelapse/files/13850553/klippy-timelapse.log)

zellneralex, 2024-01-09:
Sorry to say, but if the plugin has issues with UPDATE_DELAYED_GCODE then I would not use it, that is a basic Klipper function

zellneralex, 2024-01-09:
I would recommend you you get in contact with the owner of klipper_tmc_autotune

closed 2024-01-09 by zellneralex

Clon1998, 2024-02-06:
Quoting the solution to the issue caused by tmc auto tune for others here:

> It conflict with moonraker-timestamp https://github.com/mainsail-crew/moonraker-timelapse
> 
> Writing the configuration after [include timelapse.cfg] will solve this problem.

https://github.com/andrewmcgr/klipper_tmc_autotune/issues/80#issuecomment-1817318315