mainsail-crew/moonraker-timelapse #149: install error

TheRustyPlayer, 2024-01-16:
**Input-**
root@F005-426C /root/moonraker-timelapse [#] make install

**Return -**
/bin/sh: bash: not found
make: *** [Makefile:20: install] Error 127

Amez91, 2024-06-08:
did you ever find a fix? i have the same

anTastico, 2024-06-16:
I am also having this issue...

Ender3v3KE Rooted with Mainsail.

BiomedicLJZ, 2024-07-29:
You can fix this installing bash using opkg as `opkg install bash` but then you encounter [this](https://github.com/mainsail-crew/moonraker-timelapse/issues/171) error

anTastico, 2024-07-29:
I got mine installed with the Creality Helper Script with no issues what so ever.

BiomedicLJZ, 2024-07-29:
> I got mine installed with the Creality Helper Script with no issues with the install.
> 
> This issue I have now is it not always rendering the videos.

i tried doing the same and it does not capture a thing, any tip?

anTastico, 2024-07-29:
If I remember correctly, all I did after the install, in moonraker.conf, comment out the following:

output_path: /usr/data/printer_data/timelapse/
frame_path: /usr/data/printer_data/frames/
ffmpeg_binary_path: /opt/bin/ffmpeg
snapshoturl: http://localhost:8080/?action=snapshot

But leave the [timelapse] as is.

Like I said though, most of the times it fails to render the timelapse. It does keep all the frames in '/tmp/timelapse' so you can ssh,  copy them and render a video yourself. They are temp files though so grab them before you do another print/shut off the printer.