mainsail-crew/moonraker-timelapse #156: Instalation error with

JHdezC, 2024-03-05:
Hi, im I am trying to install the package but I get the following error with this dependency http://deb.debian.org/debian buster. I already tried changing the DNS of my pc it remains the same, what could be happening?

mks@mkspi:~/moonraker-timelapse$ make install

Ahoi!
moonraker-timelapse install routine

        This will take some time ...
        You'll be prompted for sudo password if needed!

#################### WARNING #####################
Make sure you are not printing during install!
All related services will be stopped!
##################################################

Would you like to proceed? [Y/n]: Y
Check for dependencies to use moonraker-timelapse ...
Dependency 'moonraker' found ... [OK]
Dependency 'klipper' found ... [OK]
Installing core dependencies: 'wget' ...
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://security.debian.org buster/updates InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Hit:4 http://deb.debian.org/debian buster-backports InRelease
Ign:5 http://armbian.tnahosting.net/apt buster InRelease
Err:6 http://armbian.lv.auroradev.org/apt buster Release
  404  Not Found [IP: 23.186.113.5 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100
mks@mkspi:~/moonraker-timelapse$

Kyobinoyo, 2024-03-05:
same problem here
```
Would you like to proceed? [Y/n]: Yy
Check for dependencies to use moonraker-timelapse ...
Dependency 'moonraker' found ... [OK]
Dependency 'klipper' found ... [OK]
Installing core dependencies: 'wget' ...
[sudo] password for mks:
Get:1 http://security.debian.org buster/updates InRelease [34.8 kB]
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Get:4 http://deb.debian.org/debian buster-backports InRelease [51.4 kB]
Ign:5 http://fi.mirror.armbian.de/apt buster InRelease
Err:6 http://fi.mirror.armbian.de/apt buster Release
  404  Not Found [IP: 2a01:4f9:3b:4943::2 80]
Get:7 http://security.debian.org buster/updates/main armhf Packages [754 kB]
Get:8 http://security.debian.org buster/updates/main arm64 Packages [750 kB]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100
```

tthammy, 2024-03-06:
I have exactly the same problem and i am waiting for this problem to be fixed

jordanbelinsky, 2024-03-06:
It seems that this might be related: https://forum.openmediavault.org/index.php?thread/51654-seemingly-cannot-update-and-upgrade-anymore-due-to-repository-having-no-release/

I am having the same issue on my end. Absolutely sucks.

jordanbelinsky, 2024-03-06:
I too am having the same issue. Neptune 4. Anyone have any ideas?

nachidel, 2024-03-07:
the same for me on neptune 4 max, I'm waiting for a fix.

jordanbelinsky, 2024-03-08:
It seems that apt.armbian.com no longer is hosting buster under `/dists`. Not sure what the fix would be here.

NachoBacon8r, 2024-03-09:
Same error here. It seems the buster dist isn't there anymore:

mks@mkspi:~$ cd ~/
nsail-crew/moonraker-timelapse.gmks@mkspi:~$ git clone https://github.com/mainsail-crew/moonraker-timelapse.git
Cloning into 'moonraker-timelapse'...
remote: Enumerating objects: 544, done.
remote: Counting objects: 100% (162/162), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 544 (delta 135), reused 127 (delta 125), pack-reused 382
Receiving objects: 100% (544/544), 188.13 KiB | 253.00 KiB/s, done.
Resolving deltas: 100% (277/277), done.
mks@mkspi:~$ cd ~/moonraker-timelapse
mks@mkspi:~/moonraker-timelapse$ make install

Ahoi!
moonraker-timelapse install routine

        This will take some time ...
        You'll be prompted for sudo password if needed!

#################### WARNING #####################
Make sure you are not printing during install!
All related services will be stopped!
##################################################

Would you like to proceed? [Y/n]: Y
Check for dependencies to use moonraker-timelapse ...
Dependency 'moonraker' found ... [OK]
Dependency 'klipper' found ... [OK]
Installing core dependencies: 'wget' ...
[sudo] password for mks:
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://deb.debian.org/debian buster-backports InRelease
Hit:4 http://security.debian.org buster/updates InRelease
Ign:5 http://fi.mirror.armbian.de/apt buster InRelease
Err:6 http://fi.mirror.armbian.de/apt buster Release
  404  Not Found [IP: 2a01:4f9:3b:4943::2 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100

autoferrit, 2024-03-10:
also getting this error. is there a workaround? I am installing this on a Qidi X Smart 3

Mierscheid, 2024-03-11:
Same here! Artillery Sidewinder X4 Plus.

Mierscheid, 2024-03-11:
The spaghetti detective's call has been closed. The solution in install.sh was as follows:
sudo sed -i '/^deb http:\/\/apt.armbian.com buster main buster-utils buster-desktop/s/^/# /' /etc/apt/sources.list.d/armbian.list 2>/dev/null || true

Should it also be possible to use this as a source?

Mierscheid, 2024-03-11:
Okay, that worked like this.

sudo sed -i '/^deb http:\/\/apt.armbian.com buster main buster-utils buster-desktop/s/^/# /' /etc/apt/sources.list.d/armbian.list 2>/dev/null || true

I added with nano in /scripts/install.sh.
Between the sections "Check Non Root" and "Find SRCDIR ..."

run make install again and:

Result:
moonraker-timelapse successfully installed ...

rlebrun17, 2024-03-11:
Hello ! 

Thanks @Mierscheid thanks for the tips ! 🙏🏻
In my case it's this line I had to put

`sudo sed -i '/^deb http:\/\/apt.armbian.com buster main buster-utils buster-desktop/s/^/# /' /etc/apt/sources.list.d/armbian.list 2>/dev/null || true`

Mierscheid, 2024-03-11:
OMG! That was so corrupted by not putting it as code!

Thank you! Very thoughtful! Have changed it!

Mierscheid, 2024-03-11:
ATTENTION!
ATTENTION!
ATTENTION!

As this changes the sources list, a system update is also offered in fluidd.

Do NOT start this on a Qidi printer or similar printers with a highly customised Armbian system, otherwise you will have a brick instead of a printer! 

Actually there is no update function on the Qidis, but you never know!

IMPORTANT!
IMPORTANT!
IMPORTANT!

NachoBacon8r, 2024-03-11:
So is this just a temporary workaround? I'm looking to install on a Elegoo Neptune 4 Plus which does not currently support official Fluidd updates.

Mierscheid, 2024-03-11:
No, this is a permanent workaround, as Armbian has thrown out the old sources for Buster.

Unfortunately, I can't tell you whether the Elgoo uses a "clean" Armbian. Just please do not do a system update.

jordanbelinsky, 2024-03-13:
> I added with nano in /scripts/install.sh. Between the sections "Check Non Root" and "Find SRCDIR ..."

@Mierscheid Where is this folder/script located? I can't seem to find install.sh on elegoo ssh.

jordanbelinsky, 2024-03-19:
Thank you! @Mierscheid This worked great!

awalk57, 2024-03-19:
Thank you @Mierscheid  Your fix worked great for me.

F2g-git, 2024-04-14:
I found a kinda better solution one could try.
Comment out #sudo apt-get update && sudo apt-get install --yes "${PKGLIST}"
Maybe you are lucky and have all dependents, then this would not try to update and you wouldnt get the error. Worked for me on neptune 4 pro

vanubhav, 2024-04-15:
> I found a kinda better solution one could try. Comment out #sudo apt-get update && sudo apt-get install --yes "${PKGLIST}" Maybe you are lucky and have all dependents, then this would not try to update and you wouldnt get the error. Worked for me on neptune 4 pro

@F2g-git Could you list out the steps for a non-linux comfortable user. I'm current trying to install the network manager app to use the wifi dongle on my neptune 4 pro. When I run the sudo apt install network-manager command, I get the Could not resolve 'deb.debian.org' error.

Maki9009, 2024-04-15:
i literally can find that folder moonraker timelapse folder on fluidd but i can cd to it lol

Maki9009, 2024-04-15:
okay i added it in with nano... i just don't understand how to save it everytime i try to exit and save idk what option to choose

Maki9009, 2024-04-15:
i added it it still wont work...