mainsail-crew/moonraker-timelapse #159: Repo seems to be missing the Buster Release

NachoBacon8r, 2024-03-09:
mks@mkspi:~$ cd ~/
nsail-crew/moonraker-timelapse.gmks@mkspi:~$ git clone https://github.com/mainsail-crew/moonraker-timelapse.git
Cloning into 'moonraker-timelapse'...
remote: Enumerating objects: 544, done.
remote: Counting objects: 100% (162/162), done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 544 (delta 135), reused 127 (delta 125), pack-reused 382
Receiving objects: 100% (544/544), 188.13 KiB | 253.00 KiB/s, done.
Resolving deltas: 100% (277/277), done.
mks@mkspi:~$ cd ~/moonraker-timelapse
mks@mkspi:~/moonraker-timelapse$ make install

Ahoi!
moonraker-timelapse install routine

        This will take some time ...
        You'll be prompted for sudo password if needed!

#################### WARNING #####################
Make sure you are not printing during install!
All related services will be stopped!
##################################################

Would you like to proceed? [Y/n]: Y
Check for dependencies to use moonraker-timelapse ...
Dependency 'moonraker' found ... [OK]
Dependency 'klipper' found ... [OK]
Installing core dependencies: 'wget' ...
[sudo] password for mks:
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://deb.debian.org/debian buster-backports InRelease
Hit:4 http://security.debian.org buster/updates InRelease
Ign:5 http://fi.mirror.armbian.de/apt buster InRelease
Err:6 http://fi.mirror.armbian.de/apt buster Release
  404  Not Found [IP: 2a01:4f9:3b:4943::2 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100

autoferrit, 2024-03-10:
i am also getting this issue installing on a qidi printer.

```bash
mks@mkspi:~/moonraker-timelapse$ make install

Ahoi!
moonraker-timelapse install routine

        This will take some time ...
        You'll be prompted for sudo password if needed!

#################### WARNING #####################
Make sure you are not printing during install!
All related services will be stopped!
##################################################

Would you like to proceed? [Y/n]: Y
Check for dependencies to use moonraker-timelapse ...
Dependency 'moonraker' found ... [OK]
Dependency 'klipper' found ... [OK]
Installing core dependencies: 'wget' ...
Hit:1 http://security.debian.org buster/updates InRelease
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:4 http://deb.debian.org/debian buster-updates InRelease
Hit:5 http://deb.debian.org/debian buster-backports InRelease
Ign:3 http://fi.mirror.armbian.de/apt buster InRelease
Err:6 http://fi.mirror.armbian.de/apt buster Release
  404  Not Found [IP: 65.21.120.247 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100
```

autoferrit, 2024-03-10:
looks like this thread here has more activity https://github.com/mainsail-crew/moonraker-timelapse/issues/156

BETLOG, 2024-05-17:
https://kingroon.com/blogs/3d-print-101/klipper-timelapse-fix#:~:text=If%20you%20get%20the%20%E2%80%9Cmake%3A,need%20to%20install%20everything%20manually