mainsail-crew/moonraker-timelapse #161: Timelapse setting suddenly disappeared

justjay1986, 2024-03-29:
Running up to date on everything, setting for timelapse both in the mainsail page and UI menu. Nothing changed in Moonraker.conf, or any other file. Timelapse status on console says disabled, tried reinstalling but nothing. Unsure why it's happened. 

Timelapse Setup:
enable: False
park: False
fw_retract: False
retract: 1.0 mm speed: 15 mm/s
extrude: 1.0 mm speed: 15 mm/s
verbose: True

seb449, 2024-05-15:
Same pb, impossible no menu

justjay1986, 2024-05-17:
I had to do a complete reinstall on the Rpi and Fw for the printer to get it back

ChrisH2403, 2024-06-19:
For me it worked to simply 
1. ssh into mainsail
2. cd into ~/moonraker-timelapse
3. execute make install again

Tehvan, 2024-07-14:
> For me it worked to simply
> 
>     1. ssh into mainsail
> 
>     2. cd into ~/moonraker-timelapse
> 
>     3. execute make install again

This also worked for me.

JunctionRunner, 2025-03-26:
Having the same issue here oddly, after my sd card nuked itself, and a couple fresh reinstalls. 

The timelapse option just doesn't appear in the menus, even after ssh'ing in.

I noticed that timelapse wasn't enabled in moonraker.conf, however enabling that results in moonraker dying and being unable to connect to the web ui at all, or klipperscreen, with connection refused errno 1111.

Pretty damn frustrating... It did this on my first reinstall as well, so looks like I need to do it again and just live without timelapse.