mainsail-crew/moonraker-timelapse #165: Ultra stringing (not even looks like stringing) in park mode

xTremePower, 2024-06-09:
Hello,

I installed this cool addon on my Neptune 4 max,

 I setup everything and it work perfectly in no park mode.
 
 But If I enable the park mode (for exemple in rear left), all my test models become a heary model ....
 I copied the retractation speed lenght in the Timelaps settings, but no matter what I try, it make me the same result ...

On the picture : left normal print without park . ;;; Right : parking enabled with or without z hop ...

[image]

PS : it's PETG
PS2 : I already see this exact problem here https://www.reddit.com/r/klippers/comments/v9v7in/issues_with_klipper_timelapse_stringing_like/ 
But no one single fix found

Thanks a lot

virajmalia, 2024-07-06:
I have the same issue. I am using Elegoo Neptune 4 Plus with Klipper (OpenNeptune4). Stringing happens only when park mode is enabled for timelapse.

Timelapse gcode in Orca slicer:
```gcode
G91 ; Set relative positioning
G1 Z2 F1200 ; Move Z up a bit
G90 ; Set absolute positioning
G1 X20 Y200 F9000 ; Move to top-left of bed
G4 P250 ; Pause for 250ms
TIMELAPSE_TAKE_FRAME
G4 P250 ; Pause for 250ms
```

BenMyers84, 2024-09-09:
I have this issue as well and I've not seen anyone with a solution.

xTremePower, 2024-09-09:
I completely gave up... I tried with retraction, without. Nothing helps and it only comes from the head parking mode because my impressions without are perfect. The only "solution" I've found is to park with your head in the middle of the set If the object is big like that, it hides the misery inside the object. Otherwise I deactivate the head parking, I just leave the timelaps per layer. But it's disgusting with the head that's never in the same place on the video...

bozhich, 2024-09-22:
After doing an extensive calibration and wasting 2 days trying to debug and/or workaround this issue on my K1C...

I've tried to modify the G-Code of the 'module' with hardcoded values - issue persist
Tried to copy retraction settings from a "no stringing" g-code - issue persist

Added several "sleep/dwell" ( G4 ) commands so I can try and see "what is wrong"
I've only noticed that the 'stringing' (called branching ..?) is due to filament leaking while the extruder is in the parked position. 

I suppose some kind of 'wipe' would completely solve the issue, but I don't know how to implement such a feature

virajmalia, 2024-09-27:
@xTremePower I have a Neptune 4 Pro and had the same issue . Hopefully my steps can help fix yours as well.

When you see the oozing, it means the "retract" does not happen, leaving more filament on the print. This can be verified by moving the park location to the opposite side and you will see the "ooze" happens on the same side as the park location; to be specific - the movement of the nozzle from the print to the park location. This may change according to the model's layer variation, printer speed, slicer settings and FPS settings which decide when the print is interrupted for the frame capture in relation to the layer being printed.

### Solution/Debugging
It comes down to using the right retraction settings.

1. Follow Timelapse configuration guidelines: https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/configuration.md
2. Make sure that the Park movement speed is less than or equal to your Travel speed setting in the slicer.
3. In your slicer's printer profile, check if firmware retraction is enabled and make sure you have the same setting in the Timelapse plugin's Settings.

#### If Firmware Retraction(FW) is Enabled
**Nothing to do.**
This is the easier route if you already have firmware retraction setup correctly in your `printer.cfg`. You can check this by opening your `printer.cfg` or `user_settings.cfg` and checking for a section like this:
```
 [firmware_retraction]
retract_length: 0.8
#   The length of filament (in mm) to retract when G10 is activated,
#   and to unretract when G11 is activated (but see
#   unretract_extra_length below). The default is 0 mm.
retract_speed: 60
#   The speed of retraction, in mm/s. The default is 20 mm/s.
#unretract_extra_length: 0
#   The length (in mm) of *additional* filament to add when
#   unretracting.
unretract_speed: 45
#   The speed of unretraction, in mm/s. The default is 10 mm/s.
```
The above is from my Neptune 4 Pro.

> ⚠️Note: If you do not have this section, or your values are commented out, you are using klipper default values as mentioned here: https://www.klipper3d.org/Config_Reference.html#firmware_retraction and they may or may not be the best for your printer. You need to redo your retraction calibration by disabling FW retraction everywhere, finding the right values for your printer using the retraction tower or whichever test you like using the slicer, and then enabling FW retraction and **using those values from the tests into the klipper firmware configuration** as mentioned above. The last part in **bold** is very important.

After the above steps, I got a working timelapse and a good quality print(sorry had to fix camera position during the print):
https://youtu.be/A5sdVB5AF20

[image]
[image]

#### If Firmware Retraction is Disabled
(I have not tried this approach, so it is a theory which someone may have to verify.)
When FW retraction is disabled, your slicer decides how to perform retraction.

- Look at your slicer's printer profile and copy the retract, unretract and retract speed settings over to the Timelapse settings.

bozhich, 2024-09-27:
I've come to the same conclusion, but the "oozing" does not come from "when leaving" the print, but when returning.

If You tweak around (adding PAUSE or G4) in the gcode that makes the time lapse after/before each step, You could closely monitor when and what happens.
I've tested quite extreme settings for time lapse retraction and I've only managed to make it "good" for about 5-6 sm hight print then the issue starts to appear.
I've even tried to modify a simple print's GCode, adding 
retract -> move -> dwell -> return -> extrude -> continue (several combinations of the possible orders) 
Having the same result...

If you pikachu was higher, similar issue would've happen. You can see it starting on the left ear, in the direction where the head is parked.

As all settings work and many people report the feature working (mostly) flawlessly - I think the issue is somewhere else...

virajmalia, 2024-09-27:
The issue that @xTremePower reported in the pictures is quite severe and happens immediately off the plate without any height offset like you mentioned. I had the same issue earlier but I resolved it. Maybe you have a different issue. Do you have any pictures to shows what you mean by issue starts around 5-6 cm? If you can tell me how to reproduce the same issue on my side, then I can give it a try.

Also, I did not have to make any changes to gcode except add the `TIMELAPSE_TAKE_FRAME` macro in the slicer profile. I suggest to keep it simple and let the plugin handle the command combination.

xTremePower, 2024-09-27:
Hello, thanks but i'm not a newbie .... I own 32 fdm printers up to 1 cubic meter and print since 8 years
But I don't use FW retract, I use slicer generated retract, and i tried every possible ways to remove this stringing...
It has nothing to do with slicer settings, it's something not normal with the plugin itself, because even with my cubic meter printer with 1.0mm nozzle I can cross 700mm gaps without stringing at all in normal print, and make this mess exactly at the head parking moment only

l00sed, 2025-11-17:
I was able to resolve stringing/globbing by enabling firmware retraction. In my `printer.cfg`, I added the following:

```
[include timelapse.cfg]

# Helps reduce stringing/blobbing
[firmware_retraction]
retract_length: 6.5
retract_speed: 25
unretract_extra_length: 0 # EXTRA filament to unretract
unretract_speed: 25
```