mainsail-crew/moonraker-timelapse #166: .jpg files replaced with .html code

dome-cow, 2024-06-15:
So I've had an adventure... 
Tried to install moonraker-timelapse on klipper for my sovol svo7. Didn't work, gave the following error: (apparently normal for the svo7?) 
```
E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
make: *** [Makefile:20: install] Error 100
```
I then updated klipper via the web interface, which bricked my printer until I used git over ssh to roll it back. (apparently something that shouldn't be touched!) 
Then I followed https://youtu.be/86CFV_CbrvQ?t=648 and ignored the initial problem and "installed" successfully! (timelapse settings show up in the web interface now) 

Or so I thought. I then had to add all the needed text to moonraker.conf, printer.cfg, and add timelapse.cfg and timelapse.py manually. 

**But its still not taking any photos.** When I download a .zip drive with the "photos" that it's taken, I can't open them. I looked at them with notepad, and this is what I saw: 
```
<!DOCTYPE html>
<html lang="en">
  <head>
    <script>self["MonacoEnvironment"] = (function (paths) {
          return {
            globalAPI: false,
            getWorkerUrl : function (moduleId, label) {
              var result =  paths[label];
              if (/^((http:)|(https:)|(file:)|(\/\/))/.test(result)) {
                var currentUrl = String(window.location);
                var currentOrigin = currentUrl.substr(0, currentUrl.length - window.location.hash.length - window.location.search.length - window.location.pathname.length);
                if (result.substring(0, currentOrigin.length) !== currentOrigin) {
                  var js = '/*' + label + '*/importScripts("' + result + '");';
                  var blob = new Blob([js], { type: 'application/javascript' });
                  return URL.createObjectURL(blob);
                }
              }
              return result;
            }
          };
        })({
  "editorWorkerService": "/monacoeditorwork/editor.worker.bundle.js",
  "json": "/monacoeditorwork/json.worker.bundle.js",
  "css": "/monacoeditorwork/css.worker.bundle.js",
  "less": "/monacoeditorwork/css.worker.bundle.js",
  "scss": "/monacoeditorwork/css.worker.bundle.js"
});</script>

    <meta charset="utf-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width,initial-scale=1.0">
    <title>fluidd</title>
    <meta name="description" content="The Klipper web interface for managing your 3d printer" />

    <link rel="icon" type="image/png" sizes="32x32" href="/img/icons/favicon-32x32.png" />
    <link rel="icon" type="image/png" sizes="16x16" href="/img/icons/favicon-16x16.png" />
    <meta name="theme-color" content="#2196F3" />
    <meta name="mobile-web-app-capable" content="yes" />
    <meta name="apple-mobile-web-app-capable" content="yes" />
    <meta name="apple-mobile-web-app-status-bar-style" content="black" />
    <meta name="apple-mobile-web-app-title" content="fluidd" />
    <link rel="apple-touch-icon" sizes="180x180" href="/img/icons/apple-touch-icon-180x180.png" />
    <link rel="mask-icon" href="img/icons/safari-pinned-tab.svg" color="#000000" />
    <meta name="msapplication-TileImage" content="/img/icons/mstile-150x150.png" />
    <meta name="msapplication-TileColor" content="#000000" />

    
    <script type="module" crossorigin src="/assets/index-a8560847.js"></script>
    <link rel="stylesheet" href="/assets/index-5fa27088.css">
  <link rel="manifest" href="/manifest.webmanifest">
<!-- OctoEverywhere Injected UI --><script async crossorigin src="/oe/ui.fcccd2ff04.js"></script><link crossorigin rel="stylesheet" href="/oe/ui.fcccd2ff04.css">
</head>
  <body>
    <noscript>
      <strong>We're sorry, but fluidd doesn't work properly without JavaScript enabled. Please enable it to continue.</strong>
    </noscript>
    <div id="app"></div>
    <!-- built files will be auto injected -->
  </body>
</html> 
```

I might try uninstall timelapse and re-install via this link https://github.com/mainsail-crew/moonraker-timelapse/issues/156, I'll update if that works. Either way, this weird error has got me curious as to how in the world html code would end up in the jpg files instead of actual data... 
Any help is appreciated!

dome-cow, 2024-06-20:
I got it working now!!!! 

I had to manually add 
```
enabled: True
snapshoturl: http://172.18.0.167/webcam/?action=snapshot
```
to my moonraker.conf under the `[timelapse]` line. 
I found the url by just going to my main camera on the front page of fluidd and opening image in new tab, then copying the base url of the single snapshot!

Dakweed666, 2024-08-07:
You bloody legend, this was driving me crazy, I had exactly the same problem and just couldn't figure it out, considering my K1C was working perfectly yet my KE wasn't, I had everything right EXCEPT I never had /webcam/ in the line, cheers.