mainsail-crew/moonraker-timelapse #173: Error taking timelapse frame

BenMyers84, 2024-09-02:
Im having an issue where Fluidd is giving me the following error on each layer change. "Error taking timelapse frame"
I have an ender 3 v3 se with a rooted nebula pad that I installed moon raker-timelapse on using the helper script. I have put TIMELAPSE_TAKE_FRAME in my slicer as well, I use Orca slicer. Im not sure what to include to help diagnose the issue so I have attached screenshots. 
[image]
[image]
[image]
[image]

falo2k, 2024-10-07:
I've noticed this a lot recently.  I don't know if something has changed in one of the components, but periodically frames are failing to be fetched from the snapshopt url.

`2024-10-07 14:38:15,972 [shell_command.py:_check_proc_success()] - Command (wget http://127.0.0.1/webcam/?action=snapshot -O /tmp/timelapse/printer/frame000031.jpg) successfully finished
2024-10-07 14:38:33,634 [shell_command.py:_check_proc_success()] - Command (wget http://127.0.0.1/webcam/?action=snapshot -O /tmp/timelapse/printer/frame000032.jpg) cancelled
2024-10-07 14:38:33,635 [timelapse.py:newframe()] - getting newframe failed: wget http://127.0.0.1/webcam/?action=snapshot -O /tmp/timelapse/printer/frame000032.jpg
2024-10-07 14:38:43,666 [shell_command.py:_check_proc_success()] - Command (wget http://127.0.0.1/webcam/?action=snapshot -O /tmp/timelapse/printer/frame000032.jpg) successfully finished`

iamLazyCode, 2025-02-22:
> I'm getting this a lot at the moment. I'm using the "IP Camera" app on android with a snapshot URL being http://xxx.xxx.xx.xxx:8080/photo.jpg and the `wget` command fails very often.
> 
> Initially I thought it was because the images were too high res (4K), and it timed out before being able to load the full image and lowering the resolution helped, however just tested it now and on a 250 layer print it only managed 56 frames (most are in the beginning).
> 
> Edit: Hyperlapse works fine it seems

are you using with orca slicer if yes how did u set it up please share

sashi, 2025-05-11:
Also having the same issue with wget failing at times

Setup
TPLink C120 
Frigate & RTSP for klipper/fluidd

snapshot URL is pointed to my http stream

Hyperlapse works but layer timelapse has the error
Using orca with before layer gcode

positron48, 2025-05-21:
The same issue is occurring here:

An Android phone is running the "IP Camera" app over a Wi-Fi connection, with Klipper installed on a Raspberry Pi 4B.

Fluidd occasionally shows the error: "Error taking timelapse frame." The logs provide the following output:

```
2025-05-21 13:51:03,122 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000011.jpg) successfully finished
2025-05-21 13:53:55,410 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000012.jpg) successfully finished
2025-05-21 13:56:50,652 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000013.jpg) cancelled
2025-05-21 13:56:50,653 [timelapse.py:newframe()] - getting newframe failed: wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000013.jpg
2025-05-21 13:59:45,723 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000013.jpg) cancelled
2025-05-21 13:59:45,724 [timelapse.py:newframe()] - getting newframe failed: wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000013.jpg
2025-05-21 14:02:43,814 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.0.54:8080/photo.jpg -O /tmp/timelapse/frame000013.jpg) successfully finished
```

I suspect this may be due to an unstable Wi-Fi connection or performance issues with the IP Camera app. It might be necessary to increase the timeout for the wget command. Is that possible?

positron48, 2025-05-25:
In my case, it was a timeout – by default the code sets a timeout of 2 seconds, after which an error occurs. Raising it to 5 helped eliminate it. In a test print with 80 layers, not a single error occurred.

I’ve prepared a [PR](https://github.com/mainsail-crew/moonraker-timelapse/pull/188) to configure the timeout via the config, using the wget_timelapse parameter.

A brief guide if you don’t want to wait or deal with branches:

1. Change the timeout in [this line](https://github.com/mainsail-crew/moonraker-timelapse/blob/main/component/timelapse.py#L481) from 2. to whatever you need. In my case, 5. was sufficient:

```
cmdstatus = await scmd.run(timeout=5., verbose=False)
```

2. Restart Moonraker.