mainsail-crew/moonraker-timelapse #177: make install fails

BiloxiGeek, 2024-12-25:
Tried getting this setup and it fails:

> E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details. 
> N: Repository 'http://security.debian.org buster/updates InRelease' changed its 'Suite' value from 'oldstable' to 'oldoldstable'
> N: Repository 'http://deb.debian.org/debian buster InRelease' changed its 'Suite' value from 'oldstable' to 'oldoldstable'
> N: Repository 'http://deb.debian.org/debian buster-updates InRelease' changed its 'Suite' value from 'oldstable-updates' to 'oldoldstable-updates'
> make: *** [Makefile:20: install] Error 100

Found instructions to manually install but something still isn't right. I'm trying to install on a Sovol SV07+ and I do my slicing in cura 5.9.0.

If there's a list of what steps need to happen to get this setup I'd appreciate a pointer to it. Or a good fix for the 'make install' issue.

grahamm78, 2024-12-31:
Same issue today.   Looking at the error and install.sh, it is failing trying to download wget

repro command (taken from scripts/install.sh
sudo apt-get update && sudo apt-get install --yes wget

mks@mkspi:~/moonraker-timelapse/component$ sudo apt-get update && sudo apt-get install --yes wget
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://security.debian.org buster/updates InRelease
Ign:4 http://deb.debian.org/debian buster-backports InRelease
Err:5 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 151.101.194.132 80]
Hit:6 http://repo.jing.rocks/armbian-apt buster InRelease
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I was able to comment out the line to install wget since I already had it:

#    sudo apt-get update && sudo apt-get install --yes "${PKGLIST}"

DanubeRS, 2025-08-11:
In the interest of updating the printer, I updated the apt sources list to the archived mirror URL.

The short version of my story is that after having to order an eMMC reader and re-flash this [image](https://github.com/TomasOlsson/BrickedSv07-Sv07Plus), I am now printing again.

So, commenting out the update portion of the install script is the way to go for a simple install. I have yet to try it myself, but short of sinking some time into a complete ground-up build from the [armbian](https://www.armbian.com/mks-klipad50/) image and this [handy guide](https://torte71.github.io/InsideSovolKlipperScreen/rebuilding.html).