mainsail-crew/moonraker-timelapse #181: Error with timelapse after 1384 shots "$ SET_GCODE_VARIABLE MACRO=TIMELAPSE_TAKE_FRAME VARIABLE=takingframe VALUE=False"

misterfab31, 2025-01-27:
I experience a problem on my ideaformer ir3 v2
I installed timelapse a few days ago. It rendered well for a few prints but I noticed that message for long prints
The capture stopped working after 1384 shots I got the message on the console

$ SET_GCODE_VARIABLE MACRO=TIMELAPSE_TAKE_FRAME VARIABLE=takingframe VALUE=False

Does anyone have an idea on what can cause this?

misterfab31, 2025-02-04:
Today same problem after 1419 frames.

Comparing the size of the zip export of the pictures, could it be the issue?

Is it possible to modify the maximum memory for pictures?

[image]

crowding, 2025-08-07:
Ran into this on my QIDI Q1 Pro. When I looked in moonraker.log I saw the following:

```
2025-08-08 01:39:11,985 [timelapse.py:newframe()] - cmd: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000988.jpg
2025-08-08 01:39:12,034 [timelapse.py:release_parkedhead()] - run gcommand: SET_GCODE_VARIABLE MACRO=TIMELAPSE_TAKE_FRAME VARIABLE=takingframe VALUE=False
2025-08-08 01:39:12,046 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000988.jpg) exited with return code 3
2025-08-08 01:39:12,046 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000988.jpg
2025-08-08 01:39:12,047 [timelapse.py:notify_event()] - notify_event: {'action': 'newframe', 'status': 'error'}
```

So it seems the `wget` command is failing.
I ssh'd in and ran `df -h` and saw:

```
Filesystem      Size  Used Avail Use% Mounted on
udev            417M     0  417M   0% /dev
tmpfs            98M   11M   88M  11% /run
/dev/mmcblk1p2   29G  9.4G   19G  34% /
tmpfs           489M     0  489M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           489M     0  489M   0% /sys/fs/cgroup
tmpfs           489M  489M     0 100% /tmp   # <-------------
/dev/mmcblk1p1  256M  107M  150M  42% /boot
/dev/zram1       49M   13M   33M  29% /var/log
tmpfs            98M     0   98M   0% /run/user/1000
```

So, the timelapse frames by default are written to the  `/tmp` partition, which is backed by RAM/swap and limited to half a gig. So it just fills up. Additionally this means the timelapse frames will be lost if there is a power failure.

This can be changed by setting a value for `frame_path` under the `[timelapse]` section of moonraker.conf.